jlab-sensing/ENTS-backend #58: Explicitly install libpq >= 10 for dirtviz container

adunlop621, 2023-01-13:


jmadden173, 2023-01-15:
I just updated the description to make the logs code formatted. As part of this I'm also going to add CI tests for linux, windows, and mac so this type of thing doesn't happen again.

jmadden173, 2023-01-16:
See this other github issue for more reference: https://github.com/psycopg/psycopg2/issues/1360

From the issue it seems that we can downgrade postgresql to 13 as the debian arm images have libpq 13. The docker image uses 13.9. The postgres version on the jlab server uses 13.8 so I don't forsee an issues coming of this. If you are trying to run the `docker-compose.yml` file you may see the following error because of the saved postgresql data.

```
dirtviz-postgresql-1  | 
dirtviz-postgresql-1  | PostgreSQL Database directory appears to contain a database; Skipping initialization
dirtviz-postgresql-1  | 
dirtviz-postgresql-1  | 2023-01-16 00:08:21.060 UTC [1] FATAL:  database files are incompatible with server
dirtviz-postgresql-1  | 2023-01-16 00:08:21.060 UTC [1] DETAIL:  The data directory was initialized by PostgreSQL version 14, which is not compatible with this version 13.9 (Debian 13.9-1.pgdg110+1).
dirtviz-postgresql-1 exited with code 1
```

To fix run
```
docker compose down
docker volume rm dirtviz_postgresqldata
docker compose up --build -d
```

Which will rebuild the data dir. You will have to reimport the sample data into the db.

I don't have a M1 mac so I cannot test this, @adunlop621 can you verify this works and you don't see the error mentioned above.

jmadden173, 2023-01-17:
Verified by @adunlop621, that solution work

PR #64 (2023-01-17): Revert docker compose postgres version to 13.9
Closes #58
closed 2023-01-17 by jmadden173