jlab-sensing/ENTS-backend #82: Pods OOM with gunicorn

aaron-wu1, 2023-07-05:
On high loads of the api, gunicorn starts terminating workers. An example log that would pop up is `[WARNING] Worker with pid 55 was terminated due to signal 9`.

jmadden173, 2023-07-08:
Yeah I'm not entirely sure what is going on here. Signal 9 is a SIGKILL, so something is telling it to stop running. My intuition would be that the server is running out of resources (perhaps a percentage?) so it starts shutting down resources.

cjosephson, 2023-07-08:
Do the system logs show anything useful?

aaron-wu1, 2023-07-09:
For context, this is when I'm trying to migrate the db. I haven't checked the system logs yet and I'll look into that. I did check the pod resources and it seems like there was still available resources in the pod and tried recreating the pod by deleting all the resources it was taking up but that didn't fix it. Only after waiting for some time, it automatically sorts itself out. Not sure if this is because of the migration still trying to run in the background or some background process that cleans out the used memory?

cjosephson, 2023-07-09:
I don't know what you're using to monitor processes, but I like htop. perf
<https://www.brendangregg.com/perf.html> might also help with hunting down
the issue

aaron-wu1, 2024-02-05:
Note: migrated to AWS

closed 2024-02-05 by aaron-wu1