jlab-sensing/ENTS-backend #91: Current scaling not correct

jmadden173, 2023-07-20:
The current plot and subsequently the power plot are off by a factor of 1e-6. My first thought is that a factor of 1e-6 is being applied. Perhaps due to the previous data from the RLs being in nA. Added picture for reference from the `data-collection` branch. Maybe its fixed in `main`?

@cjosephson mentioned this in the most recent EE meeting

[image]

aaron-wu1, 2023-07-22:
I'm looking at the database now and it seems like voltages are around ~0.1-0.5 and current is ~8.0^-5. Also currently power is being calculated by voltage * current, which is why the plot is around 1.0^-6 range. Maybe either voltage or current are in the wrong units?

jmadden173, 2023-07-24:
Update title for graph scaling as on the voltage/current graphs, the axis are not centered around 0 which makes it confusing when there is 0 voltage and current. Example plot below

[image]

jmadden173, 2023-07-24:
> I'm looking at the database now and it seems like voltages are around ~0.1-0.5 and current is ~8.0^-5. Also currently power is being calculated by voltage * current, which is why the plot is around 1.0^-6 range. Maybe either voltage or current are in the wrong units?

I think that it has something to do with the scaling of the data. The voltages are in `V` and the current is in `A`, therefore the current should be multiplied by 1e6 to put it in `mV`. I don't think that is being applied here.

aaron-wu1, 2023-07-27:
Notes: Make 0s aligned, don't cut out negatives

closed 2023-08-07 by aaron-wu1