jlab-sensing/ENTS-backend #145: Add API endpoint for handling serialized data from new Soil Power Sensor

jmadden173, 2024-01-09:
Waiting for https://github.com/jlab-sensing/soil-power-sensor-firmware/pull/24 to be merged. The jist is that there's a python package that you can add to the requirements with `encode` and `decode` functions for the measurement and response. The measurement will be decoded and the response will be encoded and sent back to the device giving it the OK to clear the most recent measurement.

Link to TestPyPI for package until it is actually published:
https://test.pypi.org/project/soil-power-sensor-protobuf/

aaron-wu1, 2024-02-05:
TODO: still need to test the endpoints.
```
rl_endpoint: https://dirtviz.jlab.ucsc.edu/api/power-proto/
teros_endpoint: https://dirtviz.jlab.ucsc.edu/api/teros-proto/
sensor_endpoint: https://dirtviz.jlab.ucsc.edu/api/sensor/
```

jmadden173, 2024-07-02:
Closing this issue as it looks like it's been implemented and tested. The leaf wetness deployment is running fairly smoothly.

closed 2024-07-02 by jmadden173