jlab-sensing/ENTS-backend #150: VWC axes limits misleading

cjosephson, 2024-01-25:
[image]

Maximum VWC is like 0.6 typically, it is misleading to have this from 0-100. Please set the ylim for VWC to 0.65

aaron-wu1, 2024-02-05:
To match the steps of the axis on the left with the ones on the right, the chart defaulted to setting the range of 0.7 for VWC and 700 for EC. Wondering if this is what you were looking for or if the VWC axis should be closer to 0-0.65
Here's a screenshot of what the chart looks like: 
[image]

cjosephson, 2024-02-05:
lgtm

closed 2024-02-05 by aaron-wu1