jlab-sensing/ENTS-backend #152: make EC and VWC lines different textures

cjosephson, 2024-01-30:
[image]

Our VWC/EC plot has solid lines for both qualities. To make it easier to read, can we have dashed lines for EC, and solid for the VWC?

aaron-wu1, 2024-02-05:
Because of how tightly bunched together the data points are on the graph, the dashed line doesn't show well when zoomed out. 
Here's a screenshot of the dashed lines zoomed out (first pic) and zoomed in (second pic):
[image]
[image]

If rendering the individual data points isn't necessary/needed, I could not display them and it would look like this. (No dot for each data point)
[image]
[image]

cjosephson, 2024-02-05:
Can we sub-sample the datapoints or not display them?

aaron-wu1, 2024-02-05:
I'll have to look into that. With a quick google search, I found some plugins that downsample but I'm not sure if they can downsample as the charts zoom out. I'm assuming that we want more accuracy when zoomed in and downsampling as we zoom out.

cjosephson, 2024-02-05:
yes, exactly

closed 2024-04-17 by aaron-wu1