jlab-sensing/ENTS-backend #168: dynamic tooltip labeling for charts

aaron-wu1, 2024-02-12:
[image]

Chart labels might be easier to read if the top line (out of 2 labels eg. voltage and current) was first in the tooltip hover always rather than it statically being left is always top and right is always bottom

ay-bh, 2024-03-05:
@aaron-wu1  I would like to work on this

jmadden173, 2024-11-12:
Unclear what this would accomplish

aaron-wu1, 2024-11-16:
Small task for readability, it would be easier to identify which value corresponds to which line on the graph if the ordering of the lines on the tooltip was the same as what is being displayed on the graph itself. By ordering I mean ordered by the y-axis values.

closed 2026-03-06 by jmadden173