jlab-sensing/ENTS-backend #214: Integrate PHYTOS 31 leaf wetness sensor into OSP

aaron-wu1, 2024-04-07:


aaron-wu1, 2024-04-07:
Tagging @cjosephson, @jmadden173, @stevegtaylor

jmadden173, 2024-04-19:
*Pip List*

SPS Firmware
```
Package                    Version Editable project location
-------------------------- ------- -----------------------------------------------------------------
build                      1.0.3
nanopb                     0.4.8
packaging                  23.2
pip                        23.2.1
protobuf                   4.25.1
pyproject_hooks            1.0.0
setuptools                 65.5.0
soil-power-sensor-protobuf 0.0.1   /home/jtmadden/repos/jlab/soil-power-sensor-firmware/proto/python
```

backend
```
Package                    Version
-------------------------- --------
alembic                    1.13.1
aniso8601                  9.0.1
async-timeout              4.0.3
Authlib                    1.3.0
bcrypt                     4.1.2
blinker                    1.7.0
cachelib                   0.13.0
cachetools                 5.3.3
certifi                    2024.2.2
cffi                       1.16.0
charset-normalizer         3.3.2
click                      8.1.7
cryptography               42.0.5
Flask                      3.0.3
Flask-Bcrypt               1.0.1
Flask-Cors                 4.0.0
flask-marshmallow          1.2.1
Flask-Migrate              4.0.7
Flask-RESTful              0.3.10
Flask-Session              0.8.0
Flask-SQLAlchemy           3.1.1
google-auth                2.29.0
greenlet                   3.0.3
gunicorn                   22.0.0
idna                       3.7
itsdangerous               2.2.0
Jinja2                     3.1.3
Mako                       1.3.3
MarkupSafe                 2.1.5
marshmallow                3.21.1
marshmallow-sqlalchemy     1.0.0
msgspec                    0.18.6
numpy                      1.26.4
packaging                  24.0
pandas                     2.2.2
pip                        24.0
protobuf                   5.26.1
psycopg2-binary            2.9.5
pyasn1                     0.6.0
pyasn1_modules             0.4.0
pycparser                  2.22
PyJWT                      2.8.0
python-dateutil            3.9.0
pytz                       2024.1
redis                      5.0.3
requests                   2.31.0
rsa                        4.9
setuptools                 65.5.0
six                        1.16.0
soil-power-sensor-protobuf 2.0.0
SQLAlchemy                 2.0.29
tqdm                       4.66.2
typing_extensions          4.11.0
tzdata                     2024.1
urllib3                    2.2.1
Werkzeug                   3.0.2
wheel                      0.38.4
```

In `5.x.x` of protobuf, they must have changed the implementation of `MessageToDict()`, since the decoding is failing on the serverside.

```
backend-1  |   File "/usr/local/lib/python3.11/site-packages/soil_power_sensor_protobuf/decode.py", line 53, in decode_measurement
backend-1  |     meta_dict = MessageToDict(meas.meta, including_default_value_fields=True)
backend-1  |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
backend-1  | TypeError: MessageToDict() got an unexpected keyword argument 'including_default_value_fields'
```

On the firmware side for the `soil_power_sensor_firmware` I'll peg the version of protobuf to prevent future errors occurring.

closed 2024-04-20 by aaron-wu1