jlab-sensing/ENTS-backend #355: tweak hourly/streaming toggle

cjosephson, 2024-10-28:
Instead of having a single button that toggles between hourly and streaming....
[image]

Could we put hourly/streaming as two "boxes" next to each other, and highlight the active mode as white text on blue background?

I find it difficult to tell if the button indicates the current state, or if it means "click here to go into streaming/hourly". With two buttons displayed simultaneously, it will show both possible options and also indicate which is currently selected.

devleo10, 2025-03-24:
Hey @cjosephson @jmadden173, I have worked on this issue locally, can I get assigned to this issue?

[image]

jmadden173, 2025-03-31:
@devleo10 Can you make a PR for the implementation?

devleo10, 2025-04-01:
@jmadden173 yeah sure, on it.

devleo10, 2025-04-02:
@jmadden173 Yup, done. Kindly check it.

jmadden173, 2025-05-28:
This has been fixed.