jlab-sensing/ENTS-backend #372: Improve streaming

jmadden173, 2025-02-17:
Current implementation constantly sends HTTP to get the most recent data based on the clients time. Have had trouble in the past with timezones and streaming for that reason. Quick search shows that WebSockets or Server-Sent Events (SSE) could be better alternatives.

https://blog.pixelfreestudio.com/best-practices-for-real-time-data-streaming-in-web-apps/

devleo10, 2025-03-19:
I want to work on this issue. @jmadden173

devleo10, 2025-03-21:
@aleclevy Hi, I was analyzing the network requests and noticed that similar HTTP calls are being made with the same parameters (screenshot attached). Could you confirm if this is an unintended redundant request or if there's a specific reason for it?

If it's redundant, I’d be happy to explore optimizing it, possibly with implementing SSE. Let me know your thoughts!

[image]

aleclevy, 2025-04-04:
Really not sure why this happening. It occurs both on Streaming and Hourly. I think this would be a good topic to look into optimizing.

cjosephson, 2025-10-16:
[image]
[image]

^^ two examples of ~3 minute delays experienced by two different users on two different browsers during the demo session on 10/16