jlab-sensing/ENTS-backend #387: Dashboard page going black

jmadden173, 2025-03-04:
## Describe the Bug

I was viewing cells on dirtviz (can't remember if I was streaming), switched off the tab for a indeterminate amount of time, and when I came back the page was blank.

## To Reproduce

Outline steps to reproduce the behavior:

1. Use dirtviz to view data from cells
2. Switch tabs to another page for a while 
3. Switch back to dirtviz and observe the page is blank

## Expected Behavior

Dirtviz to not be blank and showing data

## Screenshots

Screenshot with debug terminal

[image]

## Environment

- OS: `Linux spruce 6.12.16-1-lts #1 SMP PREEMPT_DYNAMIC Fri, 21 Feb 2025 19:20:31 +0000 x86_64 GNU/Linux`
- Browser: Firefox 135.01

## Additional Context

@jlin143 also has observed this issue. Maybe he can chime in on the OS and browser he's using

jmadden173, 2025-03-18:
I confirmed on my machine `linux + firefox` that it was due to streaming.

devleo10, 2025-04-11:
The error message "TypeError: id is undefined" suggests that in some of the JavaScript code there's an attempt to access the "id" property of an object that is either "undefined" or "null". The possible reason behind it could be session expiration or some logical flaw in the frontend code.

Based on my understanding, I would like to implement better error handling in "cell.js" which will check if "res.data" is "undefined" or if it lacks "id" property before attempting to use it.

If you please can confirm that you had to re-login to the application after the screen went blank, I would like to implement a token refresh mechanism to automatically refresh the tokens before they expire. This ensures that users remain authenticated. I would use Axios interceptors to enhance session management in the application.
@jmadden173 @jlin143

jmadden173, 2025-04-21:
@devleo10 I don't need to relogin to the site. I believe that this issue is related to the current streaming implementation. Both me and Jack are on firefox and have the issue. I believe the volume of streaming requests exceeds the ones allow by firefox and it stops allowing any more requests.