jlab-sensing/ENTS-backend #390: Line ending issues with `entrypoint.sh`

jmadden173, 2025-03-11:
## Describe the Bug

Windows line endings on the backend `entrypoint.sh` make it so that the interpreter cannot find the `#!/bin/bash` executable at the start of the file.

## To Reproduce

Outline steps to reproduce the behavior:

1. Locate a windows machine
2. Run `docker compose up --build`
3. Observe the error `entrypoint.sh not found`

## Expected Behavior

The local development stack to come online

## Screenshots

N/A

## Environment

- OS: Windows 11
- Browser: N/A

## Additional Context

Looks like we can fix this issue by modifying `.gitattributes` in the repo. I linked to resources below that might be useful. I don't think we need all the files to be linux line endings, only executables/scripts that might be running in the docker container.

https://stackoverflow.com/questions/65320180/issue-with-line-endings-when-developing-on-a-windows-but-running-on-a-unix-docke
https://blog.programster.org/fixing-docker-volume-windows-line-endings-on-bash-scripts

devansh1401, 2025-03-18:
I'm working on this issue.

devansh1401, 2025-03-30:
can we close this issue @jmadden173

jmadden173, 2025-03-31:
Done, thought it automatically got closed when the PR was merged.

closed 2025-03-31 by jmadden173