jlab-sensing/ENTS-backend #395: Make the logo clickable to navigate home

devansh1401, 2025-03-16:
#### **Description**  
Currently, clicking the logo in the top left does nothing. It's a common ux pattern for the logo to link to the home page, making navigation easier.  

#### **Proposed Solution**  
- Update the logo to be a link that redirects users to the home page (`/`).  

#### **Why This Is Useful**  
- Improves navigation by allowing users to easily return to the main page.  
- Aligns with standard web usability practices.  

#### **Additional Context**  
- Tested on **Google Chrome v134.0.6998.89**  .  
- Let me know if this aligns with the project’s design philosophy—happy to contribute a fix!  

---

devansh1401, 2025-03-16:
hey @jmadden173, let me know if this looks good! happy to work on it if it's needed.

jmadden173, 2025-03-18:
Hi @devansh1401, this looks like a good UI change and fairly straightforward. It would be great if you could implement it. Ping me somewhere when you get a PR up and mention the issue.

closed 2025-03-22 by devansh1401