jlab-sensing/ENTS-backend #400: SQLAlchemy Warning: Overlapping Many-to-Many Relationship in Cell and Tag Models

devleo10, 2025-03-21:
Description:
I got a SQLAlchemy warning about the many-to-many relationship between Tag and Cell. The warning indicates a conflict because both classes are defining backref relationships mapping to the same foreign key.

Error message:
SAWarning: relationship 'Tag.cells' will copy column tag.id to column cell_tag.tag_id, which conflicts with relationship(s): 'Cell.tags' (copies tag.id to cell_tag.tag_id), 'Tag.cell' (copies tag.id to cell_tag.tag_id).

Steps to Reproduce:

1) Establish the database models for Cell, Tag, and the join table Cell_Tag.
2) Establish many-to-many relationships with db.relationship and backref.
3) Run the application or start the database.
4) Notice the warning message.

Possible Fixes:
1) Replace backref with back_populates.
2) Use the overlaps parameter to avoid the warning.
Would appreciate your feedback on this!

devleo10, 2025-03-21:
@jmadden173 Could you please take a look at this issue?

jmadden173, 2025-03-31:
Kicking it back to @aleclevy as I believe that some of the changes he made relating to cell tags.

closed 2025-04-02 by devleo10