jlab-sensing/ENTS-backend #405: Plots x-axis is not synced

jmadden173, 2025-04-02:
## Describe the Bug

The x-axis for each of the sensors plots are not consistent. I was viewing data from a ENTS board that had two distinct time period (1) measuring both power and teros12 (2) switched to only measuring terso12. I would expect that there would be empty space on the power graphs where the teros12 continued to log.

This confused me as I though something had happened and the sensor stopped uploading by looking at the end date of the last uploaded point. In reality, it continued to work after I moused over the teros12 data.

## To Reproduce

See the following link showing the issue.
https://dirtviz.jlab.ucsc.edu/dashboard?cell_id=524&startDate=2025-03-19T11:08:07.871-07:00&endDate=2025-04-02T09:08:07.872-07:00

## Expected Behavior

All plot share the same x-axis so you can view patterns in the data if one sensor was taken offline then came back online.

## Screenshots

[image]

## Environment

- OS: `Linux spruce 6.12.16-1-lts #1 SMP PREEMPT_DYNAMIC Fri, 21 Feb 2025 19:20:31 +0000 x86_64 GNU/Linux`
- Browser: Mozilla Firefox 135.0.1

## Additional Context

N/A

devansh1401, 2025-04-11:
I'm working on this issue

jmadden173, 2025-04-21:
Closing issue as it was implemented in #407

closed 2025-04-21 by jmadden173