jlab-sensing/ENTS-backend #418: Contact Page: Align Auth Logic, Form Fields, and Footer with Sitewide Standards

devansh1401, 2025-04-26:
**Problems identified:**
- The Contact page does not handle authentication state for the Nav bar, causing account/profile/logout options to display incorrectly.
- The Contact form uses raw input elements instead of MUI’s `<TextField />` components, leading to inconsistent UI/UX.
- The Footer’s alignment, spacing, and icon usage are not visually appealing or consistent with the site’s branding.

**This issue tracks the work to:**
- Pass authentication state to the Nav bar on the Contact page, so user/account options only show when logged in.
- Refactor all Contact form fields to use MUI’s `<TextField />` (and `<TextField multiline />` for the message textarea) for a consistent, accessible look.
- Redesign the Footer for better alignment, spacing, and clarity, including:
  - Consistent use of icons (e.g., Email, GitHub).
  - Proper link styling and spacing.
  - Responsive, visually sharp, and minimal design.
- Ensure all code is linted, formatted, and tested according to project standards.

[image]

**Issue** : Account options incorrectly displayed when user is not logged in

closed 2025-04-28 by aleclevy

aleclevy, 2025-04-28:
Can you add this fix for authentication states to our map page? If so, feel free to just make a PR for it. Great code btw

devansh1401, 2025-04-29:
> Can you add this fix for authentication states to our map page? If so, feel free to just make a PR for it. Great code btw

Done! #422