jlab-sensing/ENTS-backend #498: Fail nicely when failing to decode measurements

jmadden173, 2025-08-20:
Likely add a exception for failed decodes that returns the error with a message indicating what happened.

API Logs Below:
```
August 20, 2025 at 15:21
	
[2025-08-20 22:21:35,770] ERROR in app: Exception on /api/sensor/ [POST]
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
Traceback (most recent call last):
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 917, in full_dispatch_request
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
rv = self.dispatch_request()
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/usr/local/lib/python3.13/site-packages/flask/app.py", line 902, in dispatch_request
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args) # type: ignore[no-any-return]
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/usr/local/lib/python3.13/site-packages/flask_restful/__init__.py", line 489, in wrapper
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
resp = resource(*args, **kwargs)
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/usr/local/lib/python3.13/site-packages/flask/views.py", line 110, in view
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
return current_app.ensure_sync(self.dispatch_request)(**kwargs) # type: ignore[no-any-return]
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/usr/local/lib/python3.13/site-packages/flask_restful/__init__.py", line 604, in dispatch_request
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
resp = meth(*args, **kwargs)
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/api/resources/sensor_data.py", line 87, in post
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
resp = self.handle_binary(data)
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/api/resources/sensor_data.py", line 146, in handle_binary
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
resp = process_measurement(data)
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/api/resources/util.py", line 33, in process_measurement
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
meas = decode_measurement(data, raw=False)
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
File "/usr/local/lib/python3.13/site-packages/ents/proto/decode.py", line 58, in decode_measurement
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
raise KeyError("Measurement missing data")
	
[347a356211af4da2b54c7fbe697151ef](https://us-west-2.console.aws.amazon.com/ecs/v2/clusters/dirtviz-cluster/services/dirtviz-service-api/tasks/347a356211af4da2b54c7fbe697151ef?region=us-west-2)
	
dirtviz-api
August 20, 2025 at 15:21
	
KeyError: 'Measurement missing data'
```

jmadden173, 2026-03-06:
Need to double check if this has been fixed.

devleo10, 2026-03-25:
@jmadden173 I've checked this issue still exists.

[image]