jlab-sensing/ENTS-backend #512: Accept semicolons for logger registration fields

jmadden173, 2025-09-03:
The logger registration form only accepts strings for App Key, MAC, Join EUI. For ease of use we should make it accept strings with `:` so that it can be directly copy and pasted from the terminal. The following is what is outputted from the LoRaWAN node that is directly copied into the registration form.

```
###### AppKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### DevEUI:      00:80:E1:15:06:09:83:82
###### AppEUI:      01:01:01:01:01:01:01:01
###### DevAddr:     06:09:83:82
```

jmadden173, 2025-09-03:
Also after registering it shows that the DevEUI was not provided.

[image]

aleclevy, 2025-11-20:
https://mui.com/material-ui/react-text-field/#integration-with-3rd-party-input-libraries

nkaplan271, 2025-11-25:
This is a good example to follow. FormattedInputs() is the main function: 

```
import * as React from 'react';
import PropTypes from 'prop-types';
import { IMaskInput } from 'react-imask';
import { NumericFormat } from 'react-number-format';
import Stack from '@mui/material/Stack';
import Input from '@mui/material/Input';
import InputLabel from '@mui/material/InputLabel';
import TextField from '@mui/material/TextField';
import FormControl from '@mui/material/FormControl';

const TextMaskCustom = React.forwardRef(function TextMaskCustom(props, ref) {
  const { onChange, ...other } = props;
  return (
    <IMaskInput
      {...other}
      mask="(#00) 000-0000"
      definitions={{
        '#': /[1-9]/,
      }}
      inputRef={ref}
      onAccept={(value) => onChange({ target: { name: props.name, value } })}
      overwrite
    />
  );
});

TextMaskCustom.propTypes = {
  name: PropTypes.string.isRequired,
  onChange: PropTypes.func.isRequired,
};

export default function FormattedInputs() {
  const handleChange = (event) => {
    setValues({
      ...values,
      [event.target.name]: event.target.value,
    });
  };

  return (
    <Stack direction="row" spacing={2}>
      <FormControl variant="standard">
        <InputLabel htmlFor="formatted-text-mask-input">react-imask</InputLabel>
        <Input
          onChange={handleChange}
          name="textmask"
          id="formatted-text-mask-input"
          inputComponent={TextMaskCustom}
        />
      </FormControl>

    </Stack>
  );
}
```